mebitek/performer #43: Moving steps in a sequence

BIOman13, 2024-01-25:
Hi,

My problem:
When recording live from a keyboard by example, certain steps  can be shifted.

A possible solution:
Would it be possible to program the following function,
- Select one or more steps in a sequence for then move them from left to right in the same sequence.

- Select step(s) (Schift key + select step)
- move the selected steps left or right (shift key + next / prev)

The same for the curve layout

Thanks,

Regards, Gil

mebitek, 2024-01-25:
what if I select step 1 and 2 and I want to move to the right if there is an enable step on 4?
what is the expeceted behaviour?

BIOman13, 2024-01-25:
If I select steps 1 and 2 and move them right to steps 3 and 4
steps 3 and 4 will be deleted and replaced by steps 1 and 2, but only after confirmation using the encoder button.

This function makes it possible to select one or more steps and move them to the right or left without affecting the existing steps, which will be overlapped.
Confirmation by pressing the encoder button will freeze the position and data of the moved steps.

mebitek, 2024-01-25:
it's a little bit tricky. I think on a solution, maybe an option to overlap or not.

BIOman13, 2024-01-25:
Effectively, my thoughts go far, I understand the complexity

mebitek, 2024-01-25:
it's not about complexity, it's more about usability

BIOman13, 2024-01-25:
Yes, It's right

BIOman13, 2024-02-03:
Hello,

Is it possible to add the following condition? 
 - The selected steps should be moved along the length of the defined sequence.

For example: if the sequence is defined as 16 steps (First step: 1 and Last step: 16) the steps to be moved are selected between steps 1 to 16.
Currently, it is done over steps 1 to 64.

Thank you for your understanding

Jil

closed 2024-02-03 by mebitek