mebitek/performer #44: Slew function

BIOman13, 2024-01-25:
Is it possible to add a SLEW to the CVs of each channel in the layout track "note" mode?

Perhaps add SLEW in the LENGTH tab by displaying a curve and allowing you to select the length of the SLEW.

SLEW is a curve logarythmic:
https://www.researchgate.net/figure/Slew-rate-control-for-Slow-Mode_fig19_237570362
Curve in bue and red

Thanks, Gil

mebitek, 2024-01-25:
do you mean the already present "slide" feature?

BIOman13, 2024-01-25:
In fact, this could be an improvement to the "SLIDE" function by adding a parameter on the duration of the "SLIDE" because currently the "SLIDE" lasts the time of the note.
If the note in longuest, the slide also.
Musically, the actuel slide doesn't always give good results.

mebitek, 2024-01-25:
Have u tried changing the slide time in the sequence page?

BIOman13, 2024-01-25:
Yes
However, the time slide acts on all notes, and it would be great if it could be defined per note.
Add a Time slide page in the NOTE tab

And then, if I currently lower the time slide to 0%, the amplitude of the note increases, for example:
- at 50%, the CV note is at 1.8V
- at 20% the CV note is at 4V
- at 80%; the note is at 0.5V
-> it's no longer the same note, the sound is weird with high notes)

I've done the test with the PAM PRO ALM and the pitch of the note doesn't change if you change the time.

BIOman13, 2024-01-26:
Hi,
Perhaps the SLIDE function could be left as it is for users who already use it.
Create a new SLEW function that can be selected and configured for each step.
This would add 2 features to the NOTE tab
- SLEW
- SLEW TIME 
- SLEW PROB, if possible

Thanks, Gil

mebitek, 2024-01-26:
due to memory limits It's very hard to impplement additional layouts, but I can see what we can do on it

BIOman13, 2024-01-26:
So perhaps we might as well delete the glide function, which musically doesn't work properly.

Gil

BIOman13, 2024-01-26:
Sorry,
SLIDE

So perhaps we might as well delete the SLIDE function, which musically doesn't work properly.