mebitek/performer #47: Improvement - shift and page

Gintetico, 2024-01-28:
Hi. To access the load/save Manu you have to press at the same time shift + menu.

Could be possible to allow also to press shift and THEN menu with 1 or 2 sec max time between the two key press?

Of course this can be applied to other menu pages that needs two key to access a page as:

Menu+t1
Menu+t2
Menu+t3
Etc

My module is vertical and with all the cable connected sometime is hard to press those little buttons at the same time. 

I hope that makes sense. 

 Thanks

mebitek, 2024-01-28:
implementing page double click to enter context menu for 2 seconds. is it ok for you?

closed 2024-01-28 by mebitek

Gintetico, 2024-01-30:
Thank you, really appreciated!

mebitek, 2024-01-30:
you welcome

Gintetico, 2024-01-30:
I wonder, if you could implemented the double click feature to enter the other submenu for all the buttons also. 

eg. double click T1 to enter Project Menu, double click T2 to enter Routing Menu, etc etc