mebitek/performer #49: Suggestion: Prevent very short output clock pulses at higher BPMs

gamecat69, 2024-01-28:
When running at higher BPMs, the clock pulses get very short and are sometimes missed by other Eurorack modules.
For example at 130 BPM, clock pulses can be as short as 2.5ms.

This is a suggestion to update Clock.pp to never allow clock pulses to drop below a certain value - 10ms would be ideal and should not be problematic even at high BPMs.

I think this can probably be done by editing line 422 in Clock.cpp.
If I understand this line correctly, the `std::max(uint32_t(1), ...)` part could allow clock pulses as short as 1ms, which is way too small.

mebitek, 2024-01-29:
done, thanks for the report

closed 2024-01-29 by mebitek

gamecat69, 2024-01-29:
Awesome, thanks!

gamecat69, 2024-01-29:
I have just tested it and it works perfectly. Thanks for addressing this so quickly.

mebitek, 2024-01-29:
perfect!!! if you want to test the other feature too, should be helpful

- Moving steps in a sequence
- INIT by step selected
- smart cycling on patter follow modes (check if launchpad is connected)
- Show launchpad settings only when a launchpad is connected
- Apply random for selected steps only
- double click page to enter context menu for 2 seconds
- Prevent very short output clock pulses at higher BPMs3
- Undo function (alt+s7)
- Curve mode backward run modes play reverse playback
- Bypass the Voltage Table in specific steps of the sequence

gamecat69, 2024-01-29:
Sure, I would be happy to test.
Please can you point me at the relevant places in the manual for these so
my tests are thorough - or perhaps provide a simple overview of each?

Kind Regards,
Nik

mebitek, 2024-01-29:
in the gh-page_feature_48 branch you will find the documentation overview for this feature.