mebitek/performer #52: Scales no longer working

gulltronica, 2024-01-30:
I've just updated from v0.1.44 to v0.1.4.48 and scales are no longer working at all as far as I can tell.

Changing either the track or universal scale no longer changes all the notes in the pattern to fit within the scale.
Changing notes manually with the encoder does not confine to the selected scale.
Random notes from the generator are no longer confined to a scale either.

Not sure if something has changed in the updates I've missed or if this is a bug.

-Dan

mebitek, 2024-01-30:
are you using an old project? if yes check the additional layer in note called bypass scale. if it is == 1 the scale will be bypassed. change it to 0 to restore the old behaviour.

gulltronica, 2024-01-30:
Sorry I forgot to mention I did try the bypass scale layer, it doesn't seem to matter either way. I've also tried with a newly initialized project.

mebitek, 2024-01-30:
it's strange, cause I just tested now and all is working as expected, also with old projects

gulltronica, 2024-01-31:
So I tried downgrading to 0.1.47 but they are still not working for me, so I went back again to 0.1.46 and they are working again. Did something with scales change in 0.1.47? 
I noticed one of the notes is "scale edit: scales are changed only if the encoder is pressed" 
is this relevant? I'm not sure what it means exactly, I've tried selecting a new scale by scrolling and clicking the encoder.

mebitek, 2024-01-31:
yes the scales now are changed just if you press the encoder because now you can jump up between scales instead of selecting continuosly.
however I'm not experiencing this kind of behavior. could you please explain better?

mebitek, 2024-02-02:
scale is not working for page+s14 quick edit function