mebitek/performer #55: Feature Request: Arpeggiator on CV/gate channels

XQSTKRPS, 2024-02-01:
Hi Mebitek!
First of all thanks again for your work.
I was wondering if it could be possible to implement something that in my opinion was an omission on the original firmware, a cv/gate arpeggiator like the one already present on MIDI tracks.
It would be just so much more useful to have it on CV, to start ideas quicker , to jam and also to perform live.
Would be cool to have different scales and rates similar to the Ableton stock Arpeggiator.
Let me know your thoughts.
Thank you
Matteo

mebitek, 2024-02-01:
could you please explain how it should works?

glibersat, 2024-02-02:
@mebitek I can backup for the behaviour when needed, let me know if you start that so we have a chat about it :)

glibersat, 2024-02-03:
Here's my take on an initial spec @mebitek 

### Parameters that should be easily changed
 - Time division (from 1/4 to 1/32T) + Rate -> **(from the current track config, nothing to add)**
 - Pattern (up, down, up up, down down, up/down, down/up, incr, decr, random). If possible, execution order is nice too but a bit harder to implement -> **(Some of them already exist in the code)**
 - **NEW** Rhythm: a few presets that alter the feel of the arp. Like "black, rest, half, half, rest, half; black" for eg. User editable is a plus but clearly not required. -> **(nothing exist atm I think)**
 - **NEW** Octave: more chances to trigger higher notes to make the arp less static -> **(nothing exist neither)**
 
### Parameters that should be honored
 - Sliding is nice to be supported so you can choose between note by note or a big slides
 - Current track scale should be honored

### How to play an arp?

 - Hitting a key will start the arp in the given key until you release it (more axed towards performance). Hitting a second key should make the later take over without interrupt so you can chain arps.
 -  If sequencing, keep the standard note sequence feeling and provides a starting and ending point. For eg, starts on 2, spans to 10. Being able to edit it later is a good thing.
 - While recording, hitting the key should place the starting point, releasing it, place the ending point.
 

### Track type

I'd say, a new track type make sense since the live play is useful here. It could maybe be merged with the "piano" request (inspired by the eloquencer, see #56)  since it's probably compatible in term of UI/UX.
I would keep the classical sequencing layout but with different note input/parameters/submenus

#### Alternative approach

Keep the current standard track and open a popup to insert a parametrized ARP. The downside of that is editability and live performance mode, not possible anymore.

### External hardware support

Being able to link a MIDI keyb or CV in to trigger the ARP even when not on the current screen is probably a good idea.

### Cleanups

I'm not sure it makes sense to keep the old arpeggiator mode reserved to Midi because it would be a more restricted, duplicate of this mode. So remove it?

XQSTKRPS, 2024-03-08:
Hey @mebitek and @glibersat , I added some stuff  to the (amazing)  previous comment by Guillaume:

**Parameters that should be easily changed**

-  Time division (from 1/4 to 1/32T) + Rate -> (from the current track config, nothing to add)   **_I would have a bigger range of time division , 1/1 or 1/2 to  1/64_**

 - Pattern (up, down, up up, down down, up/down, down/up, incr, decr, random). If possible, execution order is nice too but a bit harder to implement -> (Some of them already exist in the code)   **_agree_**
 
 - NEW Rhythm: a few presets that alter the feel of the arp. Like "black, rest, half, half, rest, half; black" for eg. User editable is a plus but clearly not required. -> (nothing exist atm I think) **_very nice concept,  i  see how that would  work nicely but i  believe it has less priority in terms of features  to implement, so it would cool if Claudio has time to do it but not mandatory in my opinion_**
 
 - NEW Octave: more chances to trigger higher notes to make the arp less static -> (nothing exist neither) **_Definitely need_**
**Parameters that should be honored**
-  Sliding is nice to be supported so you can choose between note by note or a big slides
- Current track scale should be honored **_Yep, even though it might be interesting to have a scale option on the arp  menu and being able of changing it on the fly_**
How to play an arp?
-  Hitting a key will start the arp in the given root key until you release it (more axed towards performance). Hitting a second key should make the later take over without interrupt so you can chain arps. **_Second this, but when i read it, i can just see how much comfortable it would be to have a built in Elo- style keyboard too when we introduce the arp mode.  It would really be in the spirit of the performer too , super easy to improvise or come up  with ideas without taking off your hands from the module. I am a big fan of the built in keyboard, even a crappy one.I  might just open a  separate request tho_**
- 
If sequencing, keep the standard note sequence feeling and provides a starting and ending point. For eg, starts on 2, spans to 10. Being able to edit it later is a good thing. **_absolutely. might need an extra submenu_**
While recording, hitting the key should place the starting point, releasing it, place the ending point.
The hard part here: how to choose between a button being used as an input for the current editing value OR use it as a keyboard to trigger things.  **_not sure what you mean but i would say when in arp mode a key is pressed it spaws gates and midi automatically_**

Track type
I'd say, a new track type make sense since the live play is useful here. It could maybe be merged with the "piano" request (inspired by the eloquencer, see https://github.com/mebitek/performer/issues/56) since it's probably compatible in term of UI/UX.
I would keep the classical sequencing layout but with different note input/parameters/submenus **_totally second this, and yeah it was me on the keyboard request, turns out i already opened one and didnt even remember lol. I think too they would play perfectly together. I personally dont own / use midi keyboards and even though i understand the appeal of being able to input notes easily ,  id rather be  able to  do  everything on the module _**

Alternative approach
Keep the current standard track and open a popup to insert a parametrized ARP. The downside of that is editability and live performance mode, not possible anymore. **_I see what you   mean. I believe the best thing is to see what caludio thinks is better in terms of coding_**

External hardware support
Being able to link a MIDI keyb or CV in to trigger the ARP even when not on the current screen is probably a good idea. **_100%, absolutely, cv moduleted  arps, really looking forward to it_**

Cleanups
I'm not sure it makes sense to keep the old arpeggiator mode reserved to Midi because it would be a more restricted, duplicate of this mode. So remove it? **_to be honest i have never understod why the OG coder implemented it only in midi, i always thought it was a bizzarre idea for something euro. however if it doesent occupy much memory / doesent disturb other functions, i would leave it. Doesent hurt to have it there just in case someone ever uses it (?)_**