mebitek/performer #66: Feature request: logic track

XponentOneAudio, 2024-02-10:
I often find I need to use logic alongside the performer, particularly when doing drums, and wondered if it could actually get a track type of its own so can save with the pattern and not use so many external modules?  Is this feasible?

I envision selecting the input tracks via the step buttons, selectable logic types (or/xor/and etc for gates, min/max etc for cv).  Would this work with how gates etc are handled internally?

mebitek, 2024-02-10:
I think you need to define better what this kind of track should do.

XponentOneAudio, 2024-02-10:
I'm thinking a fairly straight forward logic operation on the gate outputs - so e.g. I have a kick and snare triggered from tracks 1 and 2 with lots of random triggering, I can put a logic track on track 3 which I can set to say OR and it'll output gates whenever there a gate for either kick or snare, AND and it'll give me a gate when they hit at the same time etc.

I often find an OR of all the drum gates quite helpful in the modular, for putting noise gates on the drum track etc, so being able to have multiple inputs would be very helpful.

On the cv side with min/max, tbh I'd probably use it less but is a logical extension of the feature I figure - can be used for picking the highest/lowest of two notes etc, or maybe with curve tracks to generate a third related curve from two others.

mebitek, 2024-02-10:
it's quite a bit comlicated as the new logic track should have two other inputs track and then output the computed logic for each step if I understand what you mean

closed 2024-03-14 by mebitek