mebitek/performer #69: Automatic stop play, Add EOC function

BIOman13, 2024-02-16:
Hi,

I don't know if the function really exists because I haven't found it.

Wish :
I would like, like the ELOQUENCER for example, to be able to play the PERFORMER and have it stop automatically at the end of the chosen selection.
It would also be interesting to add looping mode.
Selections could be made on :
- one step
- several steps
- a sequence
- a song or part of a song

Interest:
I'd like this function because I've just bought the MORPHAGENE and I've found a combination of patches that allows you to record samples in a chain, automatically creating splices between the samples. It's absolutely brilliant, no need to use external software to reposition the splices. It saves a huge amount of time.

Example 1:
- I press a step (note window in the PERFORMER)
- the gate sends the recording trig to the MORPHAGENE and into an envelope (EG)
- the envelope's EOC stops the MORPHAGENE recording

Example 2:
- I select 4 steps
- I start playback
- playback stops automatically at the end of the 4th step and that's fine.
However, it is impossible to use the envelope's EOC, which would be triggered at the end of each step, preventing the 4 steps from being recorded as a single sample in the MORPHAGENE.

Possibilities:
As playback stops automatically :
- for long envelopes, you can press the MORPHAGENE stop recording button
- for very short envelopes, use of the next step could serve as an EOC

Alternatively:
- Would it be possible to program an EOC that would trigger at the end of playback of the last step of a selected sequence (at the end of the gate length. This EOC could be temporarily set to output on the RST output for example (or another output), "Temporary Sample Creation". 
- Would it be possible to add a delay (time) to this EOC, to simulate a long envelope?

Thanks

mebitek, 2024-02-16:
nice idea, but why don't use curve track to do this?
or maybe u can use conditions to play a gate just one time

BIOman13, 2024-02-17:
Hi,

I'm not sure it could work using the conditions. I haven't been able to.
As the CV is associated with the GATE, it is currently impossible to record a sequence of several steps with a start and end trigger.

The idea would be to create a function: "Record sample"

See my example in the attachment (picture attached): recording a sample over 4 steps:
[image]

- select 4 steps
- define a SOC (Start Of Cycle), predelay if required by the user at the first step
- define an EOC (Start Of Cycle), delay at the end of the last step in case the envelope is longer than the duration of the gate (this can also be associated with the tail of a reverb)
- start playback of the 4 steps with the option of stopping playback at the end of the last step or looping playback of the 4 steps.

You need to be able to separate SOC from EOC because the MORPHAGENE needs:
- SOC then EOC to start and stop recording (jack in REC input)
- SOC to create a splice between each recording (jack in the SPLICE input).

Now, in recording mode, you need to associate SOC and EOC with 2 independent outputs Can CV IN 1, 2, 3, 4 be associated with CV OUT?
I don't think so...

I was thinking of
- CLK OUT for SOC
- RST OUT for EOC

But we'll lose the clock which can be used for many external modules, LFO, ext --> bad idea.

The only solution would be to assign SOC and EOC to 2 gates of the 8 available channels, which you could select.
Example: 
Sample to record: channel 4 (GATE + CV) 
SOC: channel 1 (GATE)
EOC: channel 7 (GATE)

Assigning SOC and EOC from the PERFORMER saves the complex task of connecting an envelope and allows you to control the start and end of recording.

A function that will make people happy

I'm going to try making a video by recording a series of samples with a single step from the PERFORMER

Thanks

Best, Jil

BIOman13, 2024-02-18:
Hi,

In fact, it's possible to use CURVE mode. 
SOC : Gate CURVE
EOC : CURVE, I shift the EOC for the end of the decay

However, you'll need to provide an automatic stop for playback

I'm going to make a video

Jil

BIOman13, 2024-02-19:
Hi,

In fact, it's possible to use CURVE channel and it works well for the MORPHAGENE:
- Curve Gate: SOC (Start Of Cycle) and SPLICE creation
- Curve Shape: EOC (End Of Cycle) using a curve, the EOC represented by the curve must be shifted to take account of the decay in sound (decay, reverb, delay...)

However, it would be necessary to stop playback automatically.

Jil

mebitek, 2024-02-19:
As you can assign a route to the play toggle you can control play and stop of the sequencercer.

BIOman13, 2024-02-19:
Hi,

It is possible to start and stop the PERFORMER using routing.
But this requires the use of additional external modules and the availability of CV inputs in the PERFORMER.
In my case, everything is generally hard-wired.

I think that using routing makes it difficult and complex to record samples.

If possible:
Would it be possible to add a line in the "Project" page?
- Automatic stop of playback (yes/no)
   
No: normal PERFORMER playback

Yes: stop playback at the last selected PERFORMER step
In this case:
- display the channel to be sampled (page + Step)
- select the number of steps to be played:
        either: (shift + steps selection)  --> the steps turn red
        or: (page + first step then page + last step)
- select the play of PERFORMER 
- when playback reaches the last selected step, PERFORMER playback stops

I understand that this is a complex request but it would be great to have it

Thanks, Jil

mebitek, 2024-02-20:
I've implemented a very easy solution. A project page parameter (steps to stop) that can be from 1 to 64 value. when the engine reaches that value the clock stops. hope you can find it useful

closed 2024-02-20 by mebitek

BIOman13, 2024-02-20:
Hi,
Yes, excellent ; it's perfect
Because effectively all tracks must be affected by the play/stop process

Thanks
Jil