mebitek/performer #77: REQUEST - move one step forward via CV input

deepiain, 2024-02-27:
Request for an addition for step sequencing - Is it possible a "move one step forward"  could this be assignable to trigger via a CV input? 

this would be very, very handy when i cant quite reach over to the westlicht... 

Thanks for all your work on this, it's very exciting!

deepiain, 2024-03-05:
> can you explain better the desidered behaviour?

It would be great to be able to assign "move one step forward" to
trigger as a gate on a CV input, not sure if this would need debounced, so
it just moves one step - thats beyond my knowledge.

in use, I would assign a button or pedal and have it near to my midi
keyboard so that when programming steps I do not have to return to the
Westlicht to enter a rest, to match the XoX workflow.

fingers crossed! thank you :)

mebitek, 2024-03-05:
you mean: if the sequence is playing, you are in step 2 it will go on step3, but i fyou hit the pedal you go on step 4? or you mean the rotete function?

mebitek, 2024-03-05:
so I do not understand what you mean

deepiain, 2024-03-05:
Ok - on a 303, there is a rest button. can we have one of those?

mebitek, 2024-03-05:
it is similar to the request #84 ?

deepiain, 2024-03-05:
my bad its actually called "tap (write/step)" , its the button on the
button right of the 303.

it steps through the notes.

https://www.youtube.com/watch?v=FS3ZoFRIzxQ

deepiain, 2024-03-05:
Yes, it is.

deepiain, 2024-03-07:
the gate does move the step forward but it enters a note instead of a rest

closed 2024-03-14 by mebitek