mebitek/performer #80: Repeat Function Issue - Metropolix Mode

BIOman13, 2024-02-29:
Hi,

I've created a new project and here's a problem concerning step repetitions
Video attached

X1 work as X1
X2 work as X1
X3 work as X3
X4 work as X3
X5 work as X4
X6 work as X5
X7 work as X6
X8 work as X7

Regards, Jil

https://github.com/mebitek/performer/assets/107914379/69375bea-a5a4-4c47-9f34-c0dc59e8af01

mebitek, 2024-02-29:
Thx for reporting, does it worked before this release?

BIOman13, 2024-02-29:
Yes, the "Repeat" function used to work perfectly.

I've created projects incorporating the features of previous versions
This way, when I read them, I can identify certain inconsistencies

Project with repeat function : https://www.youtube.com/watch?v=fcqx6CetSiI
I only use the X2 in this track

Jil

mebitek, 2024-02-29:
OK I'll revert a change I made. Will release soon as I collect some more bugs

gamecat69, 2024-03-01:
I then tried to recreate the issue using a Note track and I was able to reproduce the issue:
1. Init a new project
2. Layout>Change track mode to Note
3. Track>Change play mode to free
4. Enable gate on step 1
5. Steps>Set Repeats to any value on step 1
6. Press play

Result:
(number of repeats)
x1: 1
x2: 1
x3: 2
x4: 3
x5: 4
x6: 5
x7: 6
x8: 7

mebitek, 2024-03-01:
sorry but which version are you using? I pushed change to branch feature_21

gamecat69, 2024-03-01:
using the update released on Feb 29th at 7:26 AM

gamecat69, 2024-03-01:
Tested on feature branch 0.21, this appears to be fixed

closed 2024-03-04 by mebitek