mebitek/performer #84: Request - SH101/Arturia Keystep Style sequencing

kolthrast, 2024-03-01:
I am wondering whether it would be possible to implement sh101 style/Arturia Keystep sequencing in some form, either as a separate kind of track or as a short command to the regular record function?

By sh-101 style sequencing I mean starting with no steps at all, pressing a record button and then either via midi or cv in enter notes or rests and by every note or rest you enter the sequence get a step longer. 

I imagine this could work in two ways either as a new kind of track as you generally kind of want to start with no steps/1 step and the regular tracks generally start with 16.

1. Separate kind of track: If it is done as a new kind of track I am thinking you just use the regular record button to start entering notes and when the sequencer receives a midi note or when the cv on cv in changes by at least a semitone it enters that note as a new step and moves on to the next one. If you want to enter a rest perhaps the "Next" button on the sequencer could be used for that. And if you want to create a tie, you just hold the note and press next if using midi, and perhaps press "shift" + "next" if using cv.

2. On a regular note track: Perhaps a short command could be created, I don't know that the restrictions here are , but perhaps "page" + "Shift" + "play"  or "page" + double tap "play" (or "page" + "S1" + "Play" or something like that if you want to avoid someone accidentally entering it) or something like that could enter this kind of recording instead of the normal kind. By doing that the sequence resets to having no steps/1 step and you can start entering notes and rests in the same way as above.

Thanks again for all the new functions!
/Magnus

deepiain, 2024-03-01:
Even a simple  “rest “ or “move one ste”  shortcut to implement thisI’d love this too Sent from my iPhone

mebitek, 2024-03-02:
have you look at this part of the manual?

Recording
Instead of entering note sequences one step at a time, an external MIDI keyboard can be used to speed up the process. There are two ways for recording sequences. First, sequences can be recorded live by just playing the keyboard while the sequencer is running. Secondly, step recording allows to quickly enter one note after the other using a keyboard, either when the sequencer is running or stopped. The recording mode is selected on the [Project](https://mebitek.github.io/performer/manual/?fbclid=IwAR2TSHH8D9bWoSMjLLN9zoD9pNV31jpICKs7gyQt2VWMJ9MX5nHI6Sc_-6M#pages-project) page. Recording is armed and disarmed using PAGE + PLAY.

deepiain, 2024-03-02:
Ok and how do you enter a rest? Sent from my iPhone

mebitek, 2024-03-02:
I do not know: always from the manual:

The Step Record mode is used to enter notes one step at a time. When recording is armed, playing notes on a keyboard simply sets one step after the other in the active note sequence. Slides can be recorded by moving the pitch bend control while holding a note. Ties can be recorded by moving the modulation control while holding a note. Recording starts at the first step and wraps around at the last step. Again, the length of the sequence has to be set in advance. While on the [Steps](https://mebitek.github.io/performer/manual/#pages-steps) page and the Note layer, S[1-16] can be used to select the current step to record from. The next step to be recorded into is marked with a circle.

mebitek, 2024-03-02:
I just tried:
1. enter step recording
2. click record
3. the selected step (1) is highlighted with a circle
4. press a note from the keyboard
5. next step (2) is higlighted
6. press step 4 
7. press note from keyboard

step 3 have a rest

kolthrast, 2024-03-02:
Ok wow, sorry hadn’t checked that functionality, this thing is so deep! I built mine quite soon after its release and (obviously) still find new things. Will look into if there is any way to do a “rest” kind of need that to get that arturia keystep/sh101 workflow.

deepiain, 2024-03-02:
But this is not how x0x works.. there is a rest button so you can bang away at the note, rest, slide and accents.. it’s a workflow - if you haven’t used it then it might seem like a small thing but it spoils the flow having to go back to the westlicht from keyboard / controller if I have to pick a step. Also, what if your modular is not next to your keyboard? Would be nice to assign it.Sent from my iPhone

I just tried:

enter step recording
click record

press step 4

Right, ok so it kind of have everything already, I mean would be nice with a single “rest” button or button combo rather than moving ahead by clicking the individual steps, sounds like a small thing but you know, not having to look at the screen and find the step is nice. But perhaps I have missed that that possiblity is already there. Will go back to the manual before I talk any more!

kolthrast, 2024-03-02:
Yeah, I agree, and to have that the sequence gets longer when you enter a note. I just meant I should look into the manual a but more as I had missed bits. Those small differences make a big difference in the experience! At the moment I sometime do it with a separate arturia keystep and then record it afterwards on the westlicht, which feels like an unnecessary workaround.

mebitek, 2024-03-02:
I understand the needs for a shortcut for rests. But it will be hardware dependent. For example if you use the arturia keystep we need to checknif the rest button (stop if I remember well) will send a midi message to be catch by the performer and advance the sequence by one step. For the length of the sequence the performer has 64 step fixed legnht and we can change just the first step and the last step. So we can act as the sh101 sequencer setting the last step to one and increase the value each time a midi note has been entered (or the stop rest button). But I repeat it will be hardware dependent

kolthrast, 2024-03-02:
> I understand the needs for a shortcut for rests. But it will be hardware dependent. For example if you use the arturia keystep we need to checknif the rest button (stop if I remember well) will send [...]

Right, yeah I meant if it was possible to have a rest button or command on the westlicht, sortof like your new move step command. I mean would obviously be nice to have on the keystep but perfectly decent compromise (in my mind anyway) to have it on the westlicht. That’s why I was think if it could be like a track mode, then for instance the ”next” button could be the rest when recording.

mebitek, 2024-03-05:
I've implemented CV controllable current step recorder and added shortcuts to move one step forward and backward (shift+next, shift+prev).

closed 2024-03-07 by mebitek

deepiain, 2024-03-07:
Great will try when I get home - Can we assign those move one step forward  / back shortcuts to CV? Sent from my iPhone

Closed #84 as completed.

mebitek, 2024-03-07:
forward has been already  assigned to gate. what about backward? any idea?