mebitek/performer #85: Adding a delay to the "Steps to Stop" function

BIOman13, 2024-03-01:
Hi,
I love the STOCHASTIC apgégiateur, thank you, it's wonderful.

"Steps to Stop" function works perfectly

In the example, I have a sequence of 8 steps
I use a CURVE channel to trigger the MORPHAGEN operation.

Gate: for starting and splicing
Curve: for stopping

However, the stop must be made after the eighth step to avoid cutting the decay of the last note and the FX (delay, reverb).
In the example, the stop should be made at the beginning of step 17 in order to have a 16-step sample set to the BPM.

However, in the current configuration, recording doesn't stop.

We could stop recording manually, but we would lose the ability to SLICE automatically and have a sample set to BPM.

- Would it be possible to assign a delay defined in number of steps to the desired channels? Or define the number of playback steps for each channel.

At the very least, one channel with a delay would suffice

Thanks

Jil

mebitek, 2024-03-01:
so just set the clock to stop at 17 (not 16). u can set stop up to 64 steps

BIOman13, 2024-03-01:
I'm running a sample in stochastic mode.

I have defined 8 steps to stop the sample
If I define 17 steps, the steps in stochastic mode continue to run, whereas I only want to record 8 steps.

mebitek, 2024-03-01:
so you want to stop at step 9?

BIOman13, 2024-03-01:
For the example

I would like the STOCHASTIC to play 8 steps while the "Steps to Stop" is set to 17 steps.

steps from 1 to 8: notes
steps from 9 to 16: delay (reverb, fx, etc.)
step 17: stop

mebitek, 2024-03-01:
Ok but i dont think that we can stop just a track

mebitek, 2024-03-01:
You can engage loop and set the loop lenght

BIOman13, 2024-03-01:
the loop in STOCHASTIC mode?

I tried with 
SEQ first STEP
SEQ last STEP
But these two parameters do nothing

mebitek, 2024-03-01:
You need first to engage loop (page+s7)

BIOman13, 2024-03-01:
Oopss sorry
too much novelty   :)

Perfect
Would it be possible to link "LOOP" mode to the "STEP TO STOP" function?

for example : 
if we set 17 steps on the "STEP TO STOP" function
Automatically, the "LOOP" mode will only play one loop".

if the "STEP TO STOP" function is deactivated, the loop mode will loop continuously.

BIOman13, 2024-03-01:
Ohhh pity, we practically had the best solution

mebitek, 2024-03-01:
I can check if it is possible to stop individual tracks

mebitek, 2024-03-02:
won't fix cause we cannot stop individual tracks.

closed 2024-03-02 by mebitek

BIOman13, 2024-03-02:
OK, I understand. I'm working on another solution which involves saving a sequence generated by the STOCHASTIC channel in a NOTE channel.
Always with the aim of keeping a sequence that we like, generated by the STOCHASTIC mode.

Jil