mebitek/performer #86: Copy/paste function not working

kolthrast, 2024-03-02:
Hi Claudio,

found something I believe to be a bug. The copy and paste function of steps no longer seem to work. Never tried it in your previous version so don’t know when this started. 

When on steps page and all steps are selected I press shift + page and I can select copy on the display. I then move to steps 17-32 and press shift + page and can select paste via the display menu. It says steps pasted but the steps remain blank. Perhaps I’m doing something wrong but thought I’d bring that up!

Best,
Magnus

BIOman13, 2024-03-02:
Hi Magnus,

Warning: if you select all the steps by double-clicking Shift, you will select all the steps on all 4 pages (1-16; 17-32; 33-48; 49-64).
If you only have filled steps on the first page and none on the other pages ; then when you copy all the steps on all the pages, the "PASTE" function will paste all the steps on all 4 pages, regardless of which page you are on.

If you only want to copy certain steps from one page to another
- Select the steps by clicking on SHIFT+ the steps you wish to copy
- Double-click Shift to remove the selection of copied steps.
- Go to the page where you want to paste the steps
- Select the steps by clicking on SHIFT+ the steps you wish to paste
- And "PASTE" the steps
- Double-click Shift to remove the selection of pasted steps.

mebitek, 2024-03-02:
I confirm @BIOman13 description.

kolthrast, 2024-03-02:
Thanks all, user error! Sorry about that.

closed 2024-03-02 by mebitek

BIOman13, 2024-03-02:
No problem, we're here to help each other.