mebitek/performer #88: Save a sequence from STOCHASTIC channel to a NOTE channel

BIOman13, 2024-03-02:
Hi Claudio,

I tried another solution for recording a sample from STOCHASTIC channel

Solution:
Record a sequence from a STOCHASTIC channel to a NOTE channel

Example:
8-step sequence, no particular scale or root selection

Channel 1: STOCHASTIC
I send the :
- GATE1 to CV2 IN
- CV1 to CV1 IN

I select "REC" to record the STOCHASTIC sequence on channel 8, by example

However :
- STEPS: the 1st step is not recorded (incidentally, I've noticed that I have the same problem when I use a synchronised keyboard; the first note is never recorded)
- NOTES: the last note is wrong and not part of the sequence
- REPEAT : No recorded
- RETRIG : No recorded
- LENGHT : Yes, recorded
- SLIDE : No recorded
- COND : No cheked

I may be making a mistake, but is it possible to copy a STOCHASTIC sequence to a NOTE channel?

Jil

mebitek, 2024-03-02:
could u try recording a note track into a note track?
just to see if it works as expected.

mebitek, 2024-03-02:
anotehr thing did u try setting in project page cv/gate inout when recording?

BIOman13, 2024-03-02:
Yes, the cv/gate input is selected: CV1/CV2

Here is the result of recording from NOTE to NOTE:

STEPS: the 1st step is not recorded

NOTES: OK, but not for the 1st step, which is not recorded.
If the length is 100%, the next note is not recorded.

REPEAT: For a repeat of 3, no repeat is recorded on the note, but 3 notes are created in succession.

RETRIG: Not recorded 

LENGHT: Yes, recorded

SLIDE : No recording

COND: No verification

mebitek, 2024-03-02:
so it's pretty similar to the stochastic track recording

BIOman13, 2024-03-02:
Yes

mebitek, 2024-03-02:
ok, so the main issue is that the first note will not be recorded (general issue both for nothe and stochastic).
then could you explain what u mean with " the last note is wrong and not part of the sequence"

BIOman13, 2024-03-02:
FYI : Using a KS37 keyboard, synchronised: the first note is not recorded

BIOman13, 2024-03-02:
" the last note is wrong and not part of the sequence"
As far as I'm concerned, the last note that appears to be wrong is the SLIDE interpolation of the preceding note.
If I remove the SLIDE, the note is correct.
However, the SLIDE is not recorded.

Another point: 
The LOOP function in STOCHASTIC mode does not read SLIDE, RETRIG or REPEAT.
Is this normal?

mebitek, 2024-03-02:
Yes, when u engage the loop you take a picture of the actual stochastic sequence running. If there is a slide you can engage loop and freeze the sequence and you cannot edit anymore

BIOman13, 2024-03-02:
Okay, I understand the LOOP mode

Do you think it's possible to copy a sequence (from one channel: NOTE or STOCHASTIC) into another NOTE channel, copying all the information defined for each step?

Step, note, repeat, retrig, slide...

mebitek, 2024-03-02:
I'm working on a copy loop method

closed 2024-03-04 by mebitek