mebitek/performer #90: the first note is never recorded

BIOman13, 2024-03-03:
Hi Claudio,

I mentioned it in the previous issue (14) and I don't know if you've taken it into account
If so, please disregard this issue

When automatic recording, the first note is never recorded
I did the test with different configurations: 
- KS37 in midi IN/OUT
- PERFORMER internal sequence recording
- MPC midi IN/OUT

See video attached

https://github.com/mebitek/performer/assets/107914379/1fe5d5bd-0310-4a76-88b6-002b3706e6bb

Regards, Jil

mebitek, 2024-03-03:
i think I found the issue. are you able to build the sequencer to test it?

BIOman13, 2024-03-03:
Yes, I can check

mebitek, 2024-03-03:
wait I push on branch feature_21

mebitek, 2024-03-03:
please test note track, if it works I can check the other track modes too

BIOman13, 2024-03-03:
Yes, It's the same issue with note track

See video attached
https://github.com/mebitek/performer/assets/107914379/59af96ad-ec70-46dc-94ad-2ed8a1f2ad2f

mebitek, 2024-03-03:
did you build the right branch? and another question are you in the discord group?

BIOman13, 2024-03-03:
I send GATE/CV from track 3 to the CV IN1 and IN2 inputs
I select in  PAGE/PROJECT: CV/GATE input: CV1/CV2
I go to track 7 to record GATE/CV

SloBloLabs, 2024-03-03:
> did you build the right branch? and another question are you in the discord group?
> 
> 

@mebitek off-topic this issue, which discord you're referring to?

mebitek, 2024-03-03:
@SloBloLabs its the dedicated discord server to help me in testing and developing

closed 2024-03-04 by mebitek