mebitek/performer #96: Scales in Stochastic Mode

liamjames, 2024-03-08:
Incredible firmware. Absolutely awesome, blown away by your work.

Changing scale in Stochastic mode de-highlights notes out of scale but doesn't seem to mute them- could it be an option to quantize notes to the nearest note in key, so the mode is more performative and can be used for microtonal scales?

I understand that allowing arbitrary steps (<>12) adds a huge amount of complexity that most people won't need, so this feels like a potential compromise.

mebitek, 2024-03-08:
it is made like this by design. cause you can enter arbitrary notes to creare more "musical" progression.
I had introduced the bypass scale parameter in note track to do that and stochastic track has bypass scale enabled by default for all steps.

liamjames, 2024-03-08:
Cool, I guessed it was intentional. It does work well for equal temperament scales.

I can work around it for other tunings so not a dealbreaker for me.

Thanks again for your work - brilliant project.

mebitek, 2024-03-08:
thx, maybe in the future I can unlock the bypass scale for the stochasstic track, but glad u find a work around. could you please share with us?

liamjames, 2024-03-08:
Internal performer routing works. Stochastic track CV (chromatic) routed to transpose melody track 2 (quantized to the microtonal scale). Costs two tracks instead of one but not a problem for me personally.

mebitek, 2024-03-08:
great!!!

closed 2024-03-10 by mebitek