mebitek/performer #111: Receive program change not working

greengrooves, 2024-05-02:
Hello,
im not sure if there is an option or something that i have overlooked.
But it seems that the sequencer is not reacting to program changes.
The Maleko Integration is working and i can see the program changes in the monitor.
But nothing is happening.
Would be nice if you can have a look at it
Thank you

mebitek, 2024-05-02:
I didn't develop this part. it has been taken from jackpf branches and I have not a malekko device to test it.
for what I see you need to enable malekko integration in project page (MIDI integr. parameter)

greengrooves, 2024-05-02:
No, sorry
the malekko integration is working perfectly.
I use two performer and when i send a program changes from one performer to the other,
or for example from ableton over midi to the performer, the performer itself will not react to the program changes.
But did in the last firmwire. I can see the program change in the midi monitor of the performer but its not received by the performer.

mebitek, 2024-05-02:
OK got it. Please check the other options in project page cause the malekko integration has extended it

greengrooves, 2024-05-02:
I tested over an hour, but can't get it to work.
There was also a Midi program change receive , Song mode send pattern change, in the [v0.1.4.2-1]
and Fix wrapping program changes,Properly change pattern & edit pattern when receiving midi program ch…
and 
Forward received program change messages to output by in the [v0.1.4.2-2]firmwire.
Maybee that's missing ?

mebitek, 2024-05-03:
did you try the midi integration with value "program change" ?

mebitek, 2024-05-03:
[UPDATE.DAT.zip](https://github.com/mebitek/performer/files/15196610/UPDATE.DAT.zip)

could you please try this update?

greengrooves, 2024-05-03:
Wow, thank you very much . Program change receive now works.
I'm glad I can experiment with your firmwire now :)
It's not that important, but in v0.1.4.2-1 and v0.1.4.2-2 were a few more important changes. I don't know which of them are integrated and which are not. But they were all very essential.
For example, when the performer runs in song mode and changes a preset by itself (i.e. not Manuel), it has actually also sent a program change command. I believe that was this entry:Song mode send pattern change https://github.com/jackpf/performer/pull/25
 And in the Fill Amount display is again a bug that was actually already gone. This should be the entry.: Fix UI bug with fill amount https://github.com/jackpf/performer/pull/22.

mebitek, 2024-05-03:
the song mode is already been integrated. fill amount I just  do it.
thx for reporting

mebitek, 2024-05-03:
I've integrated all jackpf changes since malekko integration [@jackpf#17](https://github.com/jackpf/performer/commit/d72af3725fc0e093764ab626a56d6e444f2717fa)

mebitek, 2024-05-03:
I repeat it's not my code, and I have no malekko gear to test so it's a little bit difficult to solve

mebitek, 2024-05-03:
did u check the midi intrg offset value? is it correct?

greengrooves, 2024-05-03:
yeah, that is on zero. But will only shift the presets around.
Maybea you can midi monitor the midi output from the performer somewhere.
like written above when sending a pr. change manually there is ch 1 number 20 value 0 on the output(That is working)
When the song mode switches a preset there should also be  ch 1 number 20 value 0.
But its ch 1 number 10 and values are 10 and 12 (That's not working)

jackpf, 2024-05-04:
Hey 👋  I don't have so much time to work on the performer firmware these days, but I can try and take a look next week if I get a chance.
Just to sumarize:
- works: sending program changes when manually changing patterns
- doesn't work: sending program change when pattern is changed through chaining patterns in song mode
Is this correct?

greengrooves, 2024-05-04:
Yes, that's correct. That would be really great.
And there is on other little bug. It's not a big problem.
The entries for program change/ Malekko, Input (midi channel /usb) and program offset.
are sometimes not saved correctly with the project. I have to dial them in every time i power on the sequencer.
Thats all :)

mebitek, 2024-05-30:
@jackpf did you manage to work on it?

closed 2024-08-06 by github-actions[bot]