mebitek/performer #114: Confusion about the Malekko PRG change implementation

ronnyjamesdisco, 2024-05-04:
I can't figure out how the Malekko integration introduced with the 0.3.0 FW is supposed to work on the hardware side. The documentation mentions how to activate it in the Project / MIDI Integration settings but I have no idea how to get the signal to the Malekko Modules - as they want to see the signal at the CV/gate bus from the 2x6 IDC. 
Where does the Performer route the signal to? I guess we then need to break out the line from the bus to a patch panel? 

This question was also discussed in MW already
https://www.modwiggler.com/forum/viewtopic.php?p=4163380#p4163380

greengrooves, 2024-05-05:
you need a special for that. I use the Expert Sleepers select bus breakout

ronnyjamesdisco, 2024-05-05:
Thanks! I think now I understood what the select bus protocol actually is ;)

closed 2024-05-07 by mebitek

ronnyjamesdisco, 2024-05-07:
Just adding, maybe the manual could reference the need for a breakout module which connects to the MIDI out in one sentence in the appendix?