mebitek/performer #117: Curved cv input problem

greengrooves, 2024-05-15:
Hello Claudio,
unfortunately I have found another problem. I don't know if it was intentional or if there is another possibility. I used to love using multiple curved tracks and recording automations with a joystick via the curved cv input for ithe selected track. So far this has always worked for the selected and armed track. As long as you could still set the curved cv input globally on the project page.
No i set in on every curved track on the track page ( i think)
If I now record something via the curved cv input, for example to track 8 wich is armed and selected in this case the data is also recorded on all other curved cv tracks. Unfortunately not only on the selected and armed one.
So I have the same cv automation on all tracks.

mebitek, 2024-05-15:
you can select the curve cv input in the track page

greengrooves, 2024-05-15:
yeah, i know. But like i said in the old firmwire i needed one curved track input and was able to record to the selected track. So i could record something to track one or two or three or eight only by selecting them and switch record on. now it's not possible.
Now i would need 4 cv ins and 4 joysticks to be able to record to 4 separate tracks. Or dive in the menu every time and switch the vc input on and off for the track every time, for every track. but that's also not practical if you will record to several tracks after another in a jam.  Is it not possible that it behaves like  it was in all the older firmware's ? Because it was really great like it was before.

mebitek, 2024-05-15:
It was a new feature called multiple cv recorder that was requested some time ago. It was requested to be used in combination with a joystick so maybe someone is using it correctly. Just ask in the Facebook group to see if u are missing something. If not I can think an option to use both modes

greengrooves, 2024-05-15:
I'm actually a little down right now since I found out that it doesn't work anymore.
The great thing about the old method was that you could record on all 8 curved tracks with one CV source e.g. a joystick. Not together, but on each track separately. Like on a big voltage block with 64 steps.
I am very sure that it is not possible to use it in the old way the way it works at the moment :(
(tested a lot)

 It would be really great if you could somehow get it to work as it did before. That would mean a lot to me.
Thank you very much

KitKatAndy, 2024-05-19:
firmware [0.3.0](https://github.com/mebitek/performer/releases/tag/0.3.0)
Hello greengrooves,
I like to try to help you. I understand correctly, that if you record on one of your curved tracks, all other curved tracks get also over-recorded, right?
I´ve tried it and at least, if you set it up correctly, you can record one curved track after another without overwriting the one´s before recorded. I don´t know, how detailed it is described in the manual, but you can setup in each track´s page, which CV-IN 1-4 should be recorded on the specific track 1-8. So you can setup to record all 4 CV-IN simultaneously into different tracks, which is especially cool with joysticks etc. To record "in sync" into a track I use in principle the routing technic described here: https://www.youtube.com/watch?v=UA-IdePG4DY
So give it a try please.

KitKatAndy, 2024-05-19:
Also just tried simple recording into a curved track via page+record after I have configured this track to "CURVE CV INPUT": CV 1.
Other (blank) curved tracks were not also recorded.

greengrooves, 2024-05-19:
Thanks for your help.
But my problem is that the way it worked before, it was possible to record with on cv source ( Joystick, Fader, knob or whatever) and one cv input to all tracks. Not at the same time but after an other. For example track one is connected to a filter, track 2 to an fx unit and so on up to track 8. Than i have one knob connected to cv input 1 and it was possible to record to the selected track. 1 up to eight and record an individual automation to the selected track. With the new method it's not possible to record different automations to all eight tracks. You can record to 4 tracks yeah, but then you need all 4 cv ins and different cv sources. For me, the old way was one of the best features.

KitKatAndy, 2024-05-19:
Ah, I read through the hole thread now... I see your problem...
You can still record in each track individually, but it needs menu diving to setup now. Understand, that this is a no go for live.
I´m in contact with mebitek. Maybe this can be solved. Will post again.

mebitek, 2024-05-20:
added option in project page to enable/disable multi cv recording. if it has been set to off the old behaviou will be onoured.

closed 2024-05-29 by mebitek