mebitek/performer #118: Stack overflow error with infinet reboot.

itramb, 2024-05-27:
Good day,
I ran into a "stack overflow in file" error, when the module keeps rebooting infinitely.
Happens both on the 0.3.0 and 0.2.2.
I fixed it temporarily by removing the LAST.DAT file, but it is reproduced by saving the project with the same settings and rebooting the module.

The configuration that kills the module:
Create 4 user scales and change it via routing from cv1.
I control it from Metropolix encoder to change scales on both modules simultaneously.
Before adding that, my projects worked and saved without any problems.
Performer runs as slave from Bitwig via ES-9 cv.
I control Performer using Lounchpad via USB.

Please help to resolve this issue somehow, other than that Performer is just a perfect sequencer for my needs :)

mebitek, 2024-05-28:
could you please report if 0.2.1 worked wuithout issues?

itramb, 2024-05-28:
Sure, will try later today

mebitek, 2024-05-28:
[UPDATE.DAT.zip](https://github.com/mebitek/performer/files/15466232/UPDATE.DAT.zip)
could you please try this update?

itramb, 2024-05-28:
I tried and it seem to be working fine, commit do not fail with multiple tracks, no errors after saving. 
All seem working perfectly !

itramb, 2024-05-28:
Thanks for the fast fix :)
So I just need to wait for 0.3.1 update for the full features?
Or is it already added ? 
For now will stay on the version provided by you :)

mebitek, 2024-05-28:
you alrady have the 0.3.1. the one I sent to you is already the 0.3.1 that includes all the fixes in the issue list

itramb, 2024-05-28:
Oh ! Thank you ! 
I sow 2.1 when updated, my bad, it's 0.3.1 in did.
Your firmware for Performer is pure magic!

closed 2024-05-29 by mebitek