mebitek/performer #122: 0.3.1 - copy/paste not working

guitoooo, 2024-05-31:
Hi, 

I found out that copy/paste action doesn"t work. 
The screen tells it's copied and pasted but no modifications on the concerned steps.

BIOman13, 2024-06-01:
Hi,

No issues for me but
Warning: if you select all the steps by double-clicking Shift, you will select all the steps on all 4 pages (1-16; 17-32; 33-48; 49-64).
If you only have filled steps on the first page and none on the other pages ; then when you copy all the steps on all the pages, the "PASTE" function will paste all the steps on all 4 pages, regardless of which page you are on.

If you only want to copy certain steps from one page to another

Select the steps by clicking on SHIFT+ the steps you wish to copy
Double-click Shift to remove the selection of copied steps.
Go to the page where you want to paste the steps
Select the steps by clicking on SHIFT+ the steps you wish to paste
And "PASTE" the steps
Double-click Shift to remove the selection of pasted steps.

Please let me know if this solves your problem
Jil

guitoooo, 2024-06-01:
Thanks a lot Jil for your very useful tips 👌

I tried again and the copy paste works fine.. also the init.. I might have not restarted the performer after upgrading the firmware.

Closing this non-issue 

Thanks!

closed 2024-06-01 by guitoooo