mebitek/performer #132: when pressing some steps (15-16?) reboots the module

onoma2, 2024-08-06:
Hello, i've just discovered this amazing fork, but during 2 hours i've spent with it it has rebooted while i was editing some parameters 3 times. will try to reproduce more specific button combos

closed 2024-08-27 by github-actions[bot]

joeljacques, 2024-10-17:
So has development on this stopped? I also experience occasional crashes. On different occcasions, but mostly when rotating sequences, while a bunch of other stuff is running as well. I actually want to play live with it next year, but I think I'll have to revert to the original stable firmware if this is sundowned.

VinxScorza, 2024-10-18:
I had a couple of crashes too. I think it happened when I was rotating the sequence backward (SHIFT+PREV). I’ll look into it more... This is not a problem when I am live, because in my last set I don't need to rotate sequences, but in general I do, so it is a bit annoying when the reboot happens and maybe I haven't saved yet...